gcormier/megadesk #187: Pinout of the buttons

makepaddev, 2025-06-30:
Hi not an issue, just a question:
Just ordered 2 on tindie, however I dont have the original controller anymore (it didn't like water) so i have to make 2 pushbuttons myself and wire it to the back of the ribbon connector, this is no problem except i would need the pinout. 

Would be awesome if you have a schematic or some way to count out how to wire it.

Thanks for your help!

gcormier, 2025-06-30:
Please email me via Tindie, I can add some repair options if you'd like to your order for a few. Otherwise the pinout is in the schematic on the releases page.

closed 2025-06-30 by gcormier

makepaddev, 2025-06-30:
[image]

gcormier, 2025-06-30:
man tindie is so pathetic sometimes. I'll shoot you an email

Check out https://github.com/gcormier/megadesk/discussions/93